ioos/glider-dac #47: SOS GetCaps has no time bounds for UW130-20150309T2005

kwilcox, 2015-05-14:
http://data.ioos.us/gliders/thredds/sos/deployments/mbari/UW130-20150309T2005/UW130-20150309T2005.nc3.nc?service=SOS&request=GetCapabilities&AcceptVersions=1.0.0

Bobfrat, 2017-08-17:
Newer datasets seem ok: https://data.ioos.us/gliders//thredds/sos/deployments/drudnick/sp007-20170427T1652/sp007-20170427T1652.nc3.nc

Need to create list of datasets that this is still an issue and we can contact the providers.

benjwadams, 2017-10-25:
This is rather strange because the OPeNDAP endpoint shows that the geospatial and time bounds related global attributes are present.  https://data.ioos.us/gliders//thredds/dodsC/deployments/mbari/UW130-20150309T2005/UW130-20150309T2005.nc3.nc.html

@cheryldmorse: any ideas here?

benjwadams, 2020-01-22:
Time bounds now show up in the SOS XML.  Closing.

closed 2020-01-22 by benjwadams