ioos/glider-dac #49: How to tell if an aggregation was updated

kwilcox, 2015-05-16:
I've found it difficult to determine when data from a glider deployment has became available or been updated.  The individual dive files are not available for download (that I could find) and scraping each deployment page looking for new or changed files seemed hacky and prone to break. I fell back to querying the DAP time variable for new timesteps. Not a terrible solution but it does not allow me to determine what else, if anything, has changed in the aggregation since the last time I checked (for example: an old dive file was manually corrected and re-uploaded).

This has been an issue with the DAP spec for as long as it has been around (no checksums).  Since the glider-dac already does some magic before aggregating the dives and making them available through DAP, could a checksum be recomputed for each deployment every time it changes and added as a global attribute?  Without this I'm forced to download every glider deployment in the DAC (fairly often) just to be sure that I'm not missing any new or corrected data.  With a checksum, someone could very quickly determine if anything in the deployment has changed.

If there is a better way to do what I'm suggesting please let me know, thanks!

kwilcox, 2015-05-28:
@kerfoot @lukecampbell is this repo the active deployed glider dac or should I raise this issue elsewhere?

kerfoot, 2015-05-28:
Kyle,

This is the active repo:

https://github.com/ioos/ioosngdac

John

closed 2015-05-28 by kwilcox