ioos/glider-dac #116: Investigate feasibility of incorporating GCOOS WaveGlider data

Bobfrat, 2017-11-29:


kerfoot, 2018-05-22:
> I do not know if this was resolved but GCOOS has a glider deployed (
> http://gandalf.gcoos.org/gandalf/), why is that glider not displayed on
> the glider map (https://gliders.ioos.us/map/#)? Did we miss a crucial
> step?

Hi Felimon,

Has this glider been registered with the DAC?:

https://data.ioos.us/gliders/providers/

and have DAC compliant NetCDF files been submitted via ftp? There is a wiki
set up to explain the process:

https://github.com/ioos/ioosngdac/wiki

In particular, please take a look at this section on deployment
registration, user accounts, wmo ids, etc:

https://github.com/ioos/ioosngdac/wiki/NGDAC-NetCDF-File-Submission-Process

Please let me know if you have any additional questions or if you would
like to schedule a call to clarify.

John

DonaldMoretti, 2023-09-05:
Enhancement.  If required, please resubmit.

closed 2023-09-05 by DonaldMoretti