ioos/glider-dac #130: Check files before archiving with NCEI

Bobfrat, 2018-06-20:
Once a deployment is marked as complete and set for archive with NCEI, we need to run the dataset through a subset of the gliderDAC compliance checker plugin to verify that the files meet the metadata standards required for NCEI. If the dataset doesn't meet these requirements, mark as Do Not Archive and send an email to the provider with the compliance checker results.

First step is to ensure all the NCEI metadata requirements are captured in the GliderDAC compliance checker plugin. If not, create those checks

benjwadams, 2018-08-24:
Can we please list the exact checks needed in this issue?  I can do this fairly easily with a shell/Python script and perhaps with the new custom plugin interface @agstephens et al. have been working on.

benjwadams, 2018-11-14:
cc-plugin-glider now has functionality to check against NCEI tables:  https://github.com/ioos/cc-plugin-glider/pull/27

We need to determine how we want to display/integrate the results of the NCEI checks in the Glider DAC application.  @kerfoot, let us know what the desired behavior is.

Bobfrat, 2020-04-21:
Can we start by just running the NCEI table checks on the ERDDAP aggregated datasets and we can adjust as needed? @kerfoot can help us come up with the email contents. Email should be sent to the providers and the DAC support email address.

benjwadams, 2021-12-01:
For now this has been revised to checking standard names prior to archival to NCEI.  #891 addresses this.

benjwadams, 2022-05-11:
This has been implemented for standard names.  No email is sent, but the deployment page lists the results of the check.  Closing as implemented.

closed 2022-05-11 by benjwadams