ioos/glider-dac #134: Mark datasets as complete after 30 days

Bobfrat, 2018-10-10:
If datasets haven't updated in 30 days we should mark the dataset as complete. 

Need to add a disclaimer on the web site as well.

kwilcox, 2018-10-10:
Does marking as complete submit the data to NCEI? Instead, can you automate an email to the person who submitted it after 30 days of no data reminding them to "complete" their deployments? I need to check with PIs before completing the deployments.

Bobfrat, 2018-10-11:
It would be submitted to NCEI if the datasets were marked for submission as well.

[image]

I like the idea of email. Hopefully thats the nudge we need, but we've found this step of coming in and checking the box is burdensome to providers so were looking for ways to improve. 

@benjwadams  I assume its no problem to send off automated emails to the provider?

@kerfoot what do you think?

kerfoot, 2018-10-11:
3 thoughts:

1. We can try the email notification and see how it works.  If we're sending out an email to notify the provider, we can likely drop the number of days from 30 to 14 or so.

2. Another option to consider is to uncheck the 'Archive to NCEI' box (it's checked by default) on the deployment registration page, thus requiring the user to actively check it and add a help/info dialog explaining what it means if the 'Archive to NCEI' box is checked.

3. Lastly, NCEI archives datasets based on MD5 checksums.  So if the provider makes any changes after the dataset is archived, the MD5 will change and NCEI will re-archive.

benjwadams, 2020-01-22:
I seem to remember some objections to this behavior.  In any event, the behavior has not changed and anything marked as incomplete will stay incomplete if not manually changed.

benjwadams, 2021-12-01:
Closing due to inactivity.  Can reopen if it's desired to revisit this.

closed 2021-12-01 by benjwadams