ioos/glider-dac #156: Enable source file access for datasets

kerfoot, 2020-08-18:
I've had a couple of requests to enable direct source file access.  I would like to make this the default.  See here:

https://coastwatch.pfeg.noaa.gov/erddap/download/setupDatasetsXml.html#accessibleViaFiles

Bobfrat, 2020-08-19:
@benjwadams looks like easiest thing to do is set this in setup.xml:
`<defaultAccessibleViaFiles>true</defaultAccessibleViaFiles>`

Bobfrat, 2020-08-20:
> @benjwadams looks like easiest thing to do is set this in setup.xml:
> `<defaultAccessibleViaFiles>true</defaultAccessibleViaFiles>`

This is a new feature in v2.03 but we're on v2.02 so we'll need to upgrade or tag each dataset in datasets.xml.

benjwadams, 2020-08-26:
Implemented by #157.

closed 2020-08-26 by benjwadams