ioos/glider-dac #163: Add all user-submitted variables to ERDDAP datasets

kerfoot, 2020-09-04:
We need to ensure that all variables in the data provider NetCDFs are included in the ERDDAP datasets.  This is currently not the case.

As an example:

DAC dataset: https://gliders.ioos.us/erddap/tabledap/sylvia-20200903T1327.html

Rutgers dataset: http://slocum-data.marine.rutgers.edu/erddap/tabledap/sylvia-20200903T1327-profile-sci-rt.html

The rutgers dataset has chlorophyll a and beta_770, among others, that are not included in the DAC datasets.

benjwadams, 2020-09-04:
Currently, we pull in any variables with a standard name and any associated variables defined within the ancillary_variables attribute of the variables which have standard names.

kerfoot, 2020-09-04:
@benjwadams ok, but I think we need to expand this to add all variables.  I've had multiple conversations with data providers asking to allow flight engineering variables into the datasets.  The argument is that some of these are pertinent to the QC of the science variables.  For example, vertical_velocity and glider pitch angle are useful when working with ADCP and CTD data.

So, IMO, we should add all variables contained in the source files.  This would also render the checking of variable:ancillary_variables attribute non-essential.

Bobfrat, 2020-09-08:
[image] [Serve all variables contained in source NetCDF files (i.e.: 'instrument_*')](https://trello.com/c/X2TAUYBP/66-serve-all-variables-contained-in-source-netcdf-files-ie-instrument)

Bobfrat, 2020-11-04:
this has been deployed to prod. the wiki should be updated to inform users though

closed 2020-12-18 by benjwadams