ioos/glider-dac #205: QC methods

thogar-computer, 2023-02-09:
Morning,

we are looking to implement some NRT QC on our glider data here at British Oceanographic Data Centre. Ideally, we would like to align with the wider community so have started looking at some of the pre-existing toolboxes. 

I noticed that your QC code is pulling from [ioos/qartod](https://github.com/ioos/qartod) which it seems is now deprecated and suggests new projects use [ioos/ioos_qc](https://github.com/ioos/ioos_qc). 

Within ioos_qc there is a glider QC file but it suggests using the [argo](https://github.com/ioos/ioos_qc/blob/main/ioos_qc/argo.py) code instead. 

I guess my question is, will you be moving to those Argo checks or are you planning to stay with the older glider ones?

Cheers
Tom

ocefpaf, 2023-02-09:
> I noticed that your QC code is pulling from [ioos/qartod](https://github.com/ioos/qartod) which it seems is now deprecated and suggests new projects use [ioos/ioos_qc](https://github.com/ioos/ioos_qc).

@thogar-computer we are slowly adding more examples to help folks migrate from qartod to ioos_qc. The first one is about to land soon. See https://github.com/ioos/ioos_qc/pull/94 (feedback is welcome).

> I guess my question is, will you be moving to those Argo checks or are you planning to stay with the older glider ones?

Cannot say much about that b/c I'm not an expert on moving platforms QA/QC but we would love to move these workflows to ioos_qc and write better documentation to help folks with that task

castelao, 2023-02-10:
Hi @thogar-computer , another alternative is [CoTeDe](https://github.com/castelao/CoTeDe) used on TSGs, XBTs, CTDs, gliders, floats, ... . Some [notebooks](https://github.com/castelao/CoTeDe/tree/master/docs/notebooks), and [procedures implemented](https://github.com/castelao/CoTeDe/tree/master/cotede/qctests) that cover the most common standards.

benjwadams, 2023-05-26:
We have some code for development environments that's being developed by @leilabbb.  We'd like to move the production routines off the legacy repository, but it still does the QC for the time being.  Yes, I'd go with one of the newer libraries if you are doing QC on a greenfield project.