ioos/glider-dac #209: Navigating to nonexistent deployment throws internal server error

benjwadams, 2023-08-02:
Navigate to a nonexistent deployment, e.g. https://gliders.ioos.us/providers/users/rutgers/deployment/qqqqqqqqqqqqqqq

HTTP 404 would be expected, looks like 500 error is being raised instead, however.

benjwadams, 2023-12-05:
Issue occurred due to trying to fetch a nonexistent deployment ID within the database and then the code would crash.  Referenced PR fixes issue, but uses deployment name instead of the internal BSON ID.

ocefpaf, 2023-12-05:
I got a 400 there instead, which is OK IMO.

```python
import requests
r = requests.get("https://gliders.ioos.us/providers/users/rutgers/deployment/qqqqqqqqqqqqqqq")
print(r.status_code)
400
```

benjwadams, 2023-12-05:
Not deployed to production yet.   404 is more appropriate and it'd be much easier to remember individual deployment names rather than IDs.  I think there are additional problems with the current system we are using that I'm currently testing.

sarinamann-noaa, 2024-08-28:
will be upgrading underlying application

closed 2024-08-28 by sarinamann-noaa