ioos/glider-dac #211: Lots of stale branches

mwengren, 2023-08-14:
Hi, I clicked through some of the PR/issue traffic this morning and noticed there are [20 public branches](https://github.com/ioos/glider-dac/branches) in this repo.  It looks like most of them are stale.

If it isn't already in the migration steps, I suggest we look at cleaning up the stale public branches that are no longer needed.

ocefpaf, 2023-08-14:
Looks like @benjwadams is the only one contributing to this repo after 2020: https://github.com/ioos/glider-dac/graphs/contributors

@benjwadams, are you OK pruning some of those branches? Many looks like a 1-off contribution made from the repo itself instead of a fork.

kbailey-noaa, 2023-09-05:
@benjwadams Can you pls review this?

kbailey-noaa, 2023-09-05:
Apologies, I added Medium Priority label by accident!

closed 2023-09-13 by benjwadams