ioos/glider-dac #213: Travis-CI is no longer a reliable service

ocefpaf, 2023-08-14:
@benjwadams we should probably move https://github.com/ioos/glider-dac/blob/master/.travis.yml to a GitHub Action. I'm not sure how often that CI is run and/or if the CI config is still used.

dextrot, 2024-02-21:
Hi @ocefpaf @ocefpaf 
if this is still required, can I take up this task?

ocefpaf, 2024-02-21:
@dextrot, if you are interested in Google summer of code, this repository is not the one you are looking for. Stay tuned in the issue you commented and we'll get in touch from there.

dextrot, 2024-02-21:
> @dextrot, if you are interested in Google summer of code, this repository is not the one you are looking for. Stay tuned in the issue you commented and we'll get in touch from there.

Roger that @ocefpaf :+1: 
Actually I've set up some GitHub actions (1. one that deploys Jekyll site to GitHub pages and 2. one that sends a mail in case a Job fails), so I thought would take a crack at this task.

benjwadams, 2024-03-27:
Closed by #328.

closed 2024-03-27 by benjwadams