ioos/glider-dac #214: submodule for docs theme needs to be updated

MathewBiddle, 2023-08-15:
Need to:
```comandline
git submodule update --init
git submodule status
git submodule update --remote --merge
git submodule status
```

MathewBiddle, 2023-08-15:
The IOOS Theme isn't rendering properly:
[image]

I think this has to do with the submodules. So I added the GHA in #215 which might close this.

mwengren, 2023-08-15:
I'm not sure I'll ever understand entirely how the `git submodule` command works, but I recently updated the IOOS Jekyll [HOWTO docs](https://ioos.github.io/ioos-documentation-jekyll-skeleton/howto.html) and added a line to run only `git submodule update` without the `--remote --merge`, ie:

```
git submodule update --init
git submodule update
```

Looks like I removed the ``--remote --merge`` options from the HOWTO entirely, can't recall why now.  It seems to work differently in different scenarios of the state of the submodule (as might be expected).

closed 2023-08-15 by MathewBiddle

MathewBiddle, 2023-08-15:
It was an issue with the config https://github.com/ioos/glider-dac/pull/216/files.
I'll update the HOWTO docs to reflect this.

commit 0546256b (2023-08-15): should fix #214
commit a1939d49 (2023-08-15): Merge pull request #216 from ocefpaf/fix_config

should fix #214