ioos/glider-dac #218: user glider now has zsh as $SHELL

kerfoot, 2023-08-15:
At some point, the login shell for user glider on dac production was changed from bash to zsh.

When doing file maintenance/troubleshooting, I typically sudo to user glider and then source my .bash_profile so that I have all of my shell tools available (i.e.: vim command line editing, history, etc.). Now that the glider shell is zsh, I am unable to do this.  It has a significant impact of the efficiency with which I am able to fix things.

1. Why was the login shell changed to zsh?
2. When was the login shell changed to zsh?
3. Can we switch back to bash?
4. If no to ioos/ioosngdac#3, do I just have to learn another shell? I've been using bash for >20 years and have become very proficient in it. Having to learn a new shell is not something I want to do.

Please advise asap.

benjwadams, 2023-08-21:
This has been in place on the Glider DAC for as long as I can remember, I'm not a huge fan of using custom shells and setups on production, although Mac OS X is now shipping with `zsh` as the default shell.  I deleted .zshrc, which is why you might be seeing messages about zsh.

`chsh` will be able to change the shell for you, there may be leftover prompt pollution from the previous config.  If that's the case, check `.profile` and a number of other related dot files.

You will want to also be able to activate the virtualenvs.   There should be commands in some of the dotfiles to source virtualenvwrapper.  If not, you can always just run `source ~/.virtualenvs/gliderdac/bin/activate`

kerfoot, 2023-08-22:
I am prompted for the glider user password when trying to change shell via chsh, but do not know the password:

1. Unless you have objections, can you please change the login shell to bash?
2. If no, how should I proceed given that I do not have the glider user password?

kerfoot, 2023-08-31:
@benjwadams I am unable to use change the shell to bash as I am prompted for a password. How should I proceed?

benjwadams, 2023-08-31:
You should have `sudo` permissions, try `sudo chsh <username>`.

kerfoot, 2023-08-31:
Yep, that worked. Closing.

closed 2023-08-31 by kerfoot