ioos/glider-dac #219: Shorten title on glider-dac GH Pages site?

mwengren, 2023-08-15:
A minor annoyance, on some screen sizes, the rather long title for the GH Pages site '[IOOS National Glider Data Assembly Center Wiki](https://ioos.github.io/glider-dac/index.html)' stretches the current menu to two lines rather than one, and in the process obscures the page titles.  In my browser it looks like: 

[image]

If the title could be shortened slightly, this should improve.

ocefpaf, 2023-08-23:
Should we do `IOOS NGDAC` for the title and have the NGDAC acronym explained in the first paragraph?

kerfoot, 2023-09-07:
How about **IOOS National Glider DAC** or is that still too long?

ocefpaf, 2023-09-08:
> How about **IOOS National Glider DAC** or is that still too long?

I think that was the title used in #265 a few weeks ago.

kbailey-noaa, 2023-09-08:
@mwengren fyi, the title length was reduced to IOOS National Glider DAC. Please close if you think this is resolved.

mwengren, 2023-09-08:
Title looks good to me. I'm looking on my phone currently and the site renders very well there.

We still need to update the overall IOOS.github.io top nav/menu bar and change the name of the Glider DAC menu item there from 'IOOS Glider DAC Data Format & Guidelines' to something else.  But that's a separate issue, so I'll close this out. 

If anyone has thoughts on what to call that link, I think I mentioned it in our gdac repo email thread, or there's discussion and a list of menu changes here:

https://github.com/ioos/documentation-theme-jekyll/issues/10#issue-1603274792

closed 2023-09-08 by mwengren