ioos/glider-dac #220: glider name not parsed during deployment registration

kerfoot, 2023-08-15:
For 2 recent deployments:

- ng291-20230726T164200
- ng296-20230726T172700

was not saved to the backend db used for deployment registration. The value for the __*glider*__ field is *null*:

    {
          "archive_safe": true, 
          "attribution": null, 
          "checksum": "d41d8cd98f00b204e9800998ecf8427e", 
          "completed": false, 
          "created": 1690414295622, 
          "dap": "http://gliders.ioos.us/thredds/dodsC/deployments/navoceano/ng291-20230726T164200/ng291-20230726T164200.nc3.nc", 
          "delayed_mode": false, 
          "deployment_date": null, 
          "deployment_dir": "navoceano/ng291-20230726T164200", 
          "erddap": "http://gliders.ioos.us/erddap/tabledap/ng291-20230726T164200.html", 
          "estimated_deploy_date": null, 
          "estimated_deploy_location": null, 
          "glider_name": null, 
          "id": "64c1acd74172d80001a6fea5", 
          "iso": "http://gliders.ioos.us/erddap/tabledap/ng291-20230726T164200.iso19115", 
          "latest_file": null, 
          "latest_file_mtime": null, 
          "name": "ng291-20230726T164200", 
          "operator": null, 
          "sos": "http://gliders.ioos.us/thredds/sos/deployments/navoceano/ng291-20230726T164200/ng291-20230726T164200.nc3.nc?service=SOS&request=GetCapabilities&AcceptVersions=1.0.0", 
          "thredds": "http://gliders.ioos.us/thredds/catalog/deployments/navoceano/ng291-20230726T164200/catalog.html?dataset=deployments/navoceano/ng291-20230726T164200/ng291-20230726T164200.nc3.nc", 
          "updated": 1690817494288, 
          "username": "navoceano", 
          "wmo_id": null
        }, 
    {
          "archive_safe": true, 
          "attribution": null, 
          "checksum": "d41d8cd98f00b204e9800998ecf8427e", 
          "completed": false, 
          "created": 1690421488929, 
          "dap": "http://gliders.ioos.us/thredds/dodsC/deployments/navoceano/ng296-20230726T172700/ng296-20230726T172700.nc3.nc", 
          "delayed_mode": false, 
          "deployment_date": null, 
          "deployment_dir": "navoceano/ng296-20230726T172700", 
          "erddap": "http://gliders.ioos.us/erddap/tabledap/ng296-20230726T172700.html", 
          "estimated_deploy_date": null, 
          "estimated_deploy_location": null, 
          "glider_name": null, 
          "id": "64c1c8f04172d80001a6fea6", 
          "iso": "http://gliders.ioos.us/erddap/tabledap/ng296-20230726T172700.iso19115", 
          "latest_file": null, 
          "latest_file_mtime": null, 
          "name": "ng296-20230726T172700", 
          "operator": null, 
          "sos": "http://gliders.ioos.us/thredds/sos/deployments/navoceano/ng296-20230726T172700/ng296-20230726T172700.nc3.nc?service=SOS&request=GetCapabilities&AcceptVersions=1.0.0", 
          "thredds": "http://gliders.ioos.us/thredds/catalog/deployments/navoceano/ng296-20230726T172700/catalog.html?dataset=deployments/navoceano/ng296-20230726T172700/ng296-20230726T172700.nc3.nc", 
          "updated": 1690831905929, 
          "username": "navoceano", 
          "wmo_id": null
        }, 

There are 1654 registered deployments and these are the only 2 for which the value of __*glider_name*__ == *null*. This api end point is used to populate the [status page](https://gliders.ioos.us/status/?cat=datasets&mode=rt&type=incomplete&hrs=0) and is preventing these 2 deployments from being properly cataloged and displayed. These 2 deployments were created on July 26 and July 27, 2023.

benjwadams, 2023-09-06:
Issue likely occurred due to separate deployment creation path via watchdog script.  This was removed in 7cd361e41c4a72b248005d8a588ac6db8b48f4db and so issue may not recur due to this.

kerfoot, 2023-09-06:
@benjwadams sounds like we can close this?

kerfoot, 2023-09-06:
@benjwadams Can we close this?

Acolohan, 2023-09-07:
If its ready to be closed, go ahead and Ill move it to the "done" column

kerfoot, 2023-09-07:
Closing. Likely an outlier for the reasons @benjwadams stated above. Subsequent data set registrations have the glider stored in the db.

closed 2023-09-07 by kerfoot