ioos/glider-dac #221: Establish modifications to QC process

benjwadams, 2023-08-15:
Please list desired changes to QC process.  This does not directly include things like the documentation and migrating to the `ioos_qc` library.

So far:

1) Always apply QARTOD to variables, regardless of whether user supplied QC variables are provided.
2) Eliminate `_qc` variables which appear unused in favor of aggregate flags.

kerfoot, 2023-08-31:
One additional change:

1. Remove the 15 or so legacy variables noted in [issue #268](https://github.com/ioos/glider-dac/issues/268#issue-1867455043) unless they are included in the files submitted by the data providers.

benjwadams, 2023-09-05:
1) in original post should be split out into separate issue if it isn't already.

2) #277 covers aggregate flags,
   #278 covers legacy `*_qc` variables.
   
Would recommend closing this issue once adequately split up.

kerfoot, 2023-09-08:
I would suggest closing this as I believe it is being addressed in a series of other issues that are more specific with the overall goal of establishing an updated QC process.

kerfoot, 2023-09-13:
All issues/comments above have been addressed in smaller, more focused issues. Closing.

closed 2023-09-13 by kerfoot