ioos/glider-dac #223: Raw Data Upload Tool

kerfoot, 2023-08-15:
Raw data upload tool development and documentation.

Acolohan, 2023-10-03:
Was just reviewing the statement of work with Kathy and was wondering if there is a user name attached to this one? We can talk about it tomorrow.

kerfoot, 2023-10-03:
I think it's @kerfoot as the primary for design and development, then @nguyandy once we need it built.

Acolohan, 2023-10-03:
ok, we can talk about it more in tomorrows meeting

-- 
Very Respectfully,
LCDR Aaron D. Colohan
Executive Officer,
US IOOS Program
Personal: 804-514-7035
Work: ‪(251) 207-1020‬

"We sympathize in the great moments of history, in the great discoveries,
the great resistances, the great prosperities of men; because there law was
enacted, the sea was searched, the land was found, or the blow was struck, *for
us*, as we ourselves in that place would have done or applauded."  -
Emerson

closed 2024-03-08 by kbailey-noaa