ioos/glider-dac #224: Compliance Checking

kerfoot, 2023-08-15:
Documentation of how and when the compliance checks run and make the results of the compliance check available via an API.  This will allow administrators and data providers to receive feedback on the status of data sets and why they may not be available for NCEI archive.

benjwadams, 2024-05-09:
As I recall, compliance checks are run upon marking deployments as complete and ready for archival, provided there are actual files present.
I will verify this.

Compliance check results are stored in a JSON field within the database.

benjwadams, 2024-05-09:
We also have logic for sending emails for the compliance check reports.  It's currently unused, but could be easily reactivated.

benjwadams, 2025-11-13:
Already present in docs.

closed 2025-11-13 by benjwadams