ioos/glider-dac #229: Disassociate data.ioos.us domain with Glider DAC project

benjwadams, 2023-08-15:
Historically, https://data.ioos.us/gliders used to be the root location of most Glider DAC pages.  Nowadays, there is a HTTPS redirect to https://gliders.ioos.us.

However, FTP still can be accessed via either data.ioos.us or gliders.ioos.us .   The former use should be eliminated, as the data.ioos.us subdomain now pertains mostly to IOOS Catalog.  This is, in fact casuing issues for directory listing since firewall rules changed since IOOS Catalog resides in another AWS account and firewall rules no longer allow cross account transferring of files via `rsync`.  It's already mentioned in the FTP MOTD that users should move to gliders.ioos.us for FTP/FTPS needs.  Switching over would require updating of DNS records, which is simple, but would no longer allow data.ioos.us to function as an FTP endpoint, barring some workaround to somehow forward FTP traffic.  I'm not sure that such a solution is readily available, so the likely case is that data.ioos.us will cease to function as an FTP endpoint.

benjwadams, 2023-08-21:
@kerfoot, we'd like to cut this over soon.

kerfoot, 2023-08-31:
@benjwadams Is this complete? If so, can you close it please?

kerfoot, 2023-09-06:
Done. closing.

kerfoot, 2023-09-06:
reopening. IT discussion.

closed 2023-09-20 by benjwadams