ioos/glider-dac #230: Disable FTP in favor of FTPS

benjwadams, 2023-08-15:
FTP should be eliminated in favor of FTPS due to security concerns.  This is very easy to realize, the FTP config just needs one value changed and then application needs to be restarted and FTP will be disabled.  There is already a MOTD on the FTP indicating that users should move away from FTP, but a hard date for cutoff needs to be set and communication to providers outside of this MOTD should be established.

kerfoot, 2023-09-13:
Should be an easy fix but in the interest of hurricane season, we are deferring this until December 2023.

sarinamann-noaa, 2024-09-27:
Ben confirmed that this issue is still a priority after hurricane season

sarinamann-noaa, 2025-03-17:
@benjwadams just wanted to get this issue on the radar as we enter Q2 before hurricane season 2025.
Thanks!