ioos/glider-dac #232: Create a Robust Development Environment

DonaldMoretti, 2023-08-15:
Create an appropriate robust development environment to allow proper testing prior to production releases to ensure we do not risk our production infrastructure

kerfoot, 2023-09-14:
Can we also create the dev status page? Should be located at

https://dev-gliders.ioos.us/status

but this end point is not currently functional.

DonaldMoretti, 2023-09-20:
@benjwadams to evaluate effort for ERDDAP Upgrade

kerfoot, 2023-09-22:
I am unable to register and deployment and upload files in order to get the data set to show up on the dev ERDDAP server. I registered the deployment and dropped 5 files via the file drag and drop and I see them on the webpage (see attached screenshot). But, when I look in:
 
/data/submission/rutgers
 
That data set is not there:
 
    (gliderdac)➜  rutgers  pwd
    /data/submission/rutgers
    (gliderdac)➜  rutgers  ls -l
    total 260
    drwxr-xr-x 2 glider glider   4096 Aug 22 01:47 ng000-20230821T0000
    drwxr-xr-x 2 glider glider  57344 Jun 15 18:47 ng231-20190901T0000
    drwxr-xr-x 2 glider glider  32768 Jan 13  2023 ru29-20210908T1943
    drwxr-xr-x 2 glider glider 163840 Feb 14  2023 ru32-20210330T1448
    drwxr-xr-x 2 glider glider   4096 Feb  9  2023 ru99-20230712T0000
 
There is no directory for the ru38-20230420T1602 data set that I registered:
 
https://dev-gliders.ioos.us/providers/users/rutgers/deployment/650b483ea15ad60009e699db

When I navigate to the link above, the files have disappeared.
 
So, it seem that even though I registered the deployment, the corresponding directories on the dev server have not been created and do not exist. Not sure where the files that I dropped are actually going, but they are not going where they are supposed to as those directories do not exist:
 
    /data/submission/rutgers
    (gliderdac)➜  rutgers  ls -l
    total 260
    drwxr-xr-x 2 glider glider   4096 Aug 22 01:47 ng000-20230821T0000
    drwxr-xr-x 2 glider glider  57344 Jun 15 18:47 ng231-20190901T0000
    drwxr-xr-x 2 glider glider  32768 Jan 13  2023 ru29-20210908T1943
    drwxr-xr-x 2 glider glider 163840 Feb 14  2023 ru32-20210330T1448
    drwxr-xr-x 2 glider glider   4096 Feb  9  2023 ru99-20230712T0000
 
or:
 
    (gliderdac)➜  rutgers  pwd
    /data/data/priv_erddap/rutgers
    (gliderdac)➜  rutgers  ls -l
    total 268
    drwxr-xr-x 2 glider glider   4096 Aug 22 01:47 ng000-20230821T0000
    drwxr-xr-x 2 glider glider  65536 Jun 15 18:47 ng231-20190901T0000
    drwxr-xr-x 2 glider glider  32768 Jan 13  2023 ru29-20210908T1943
    drwxr-xr-x 2 glider glider 163840 Feb 14  2023 ru32-20210330T1448
    drwxr-xr-x 2 glider glider   4096 Feb  9  2023 ru99-20230712T0000
 
So it looks like the dev environment is not functional in terms of being able to register a deployment and submit files.

benjwadams, 2023-10-05:
Problem described immediately above by @kerfoot has been fixed, it was a misconfiguration issue.

This issue is a bit too broad, and should be broken down into actionable subparts.

benjwadams, 2024-01-23:
The branch on https://github.com/ioos/glider-dac/tree/aws_environment_sync will probably suffice for a good development environment.  We can discuss tomorrow and decide if we want to close this as completed.

PR #315 (2024-01-24): Dev environment
Allows easier creation of dev environment through Docker and pull of data from AWS S3. Fixes #232.
closed 2024-01-24 by benjwadams