ioos/glider-dac #233: NCEI Archival Package Creation is Broken

kerfoot, 2023-08-15:
The current NCEI archival package creation is poorly understood and is broken. No new packages have been created and archived by NCEI since July 2022.

# Questions
1. What is the current (broken) archival package process?
2. How is it scheduled?

# Tasks
1. Understand the current (broken) archival package creation process
2. Build an automated archival package process
3. Determine archival package creation schedule. Potentially different schedules for real-time and delayed-mode processing.
4. Provide a manual process for creating archival packages in the event the automated process breaks again.
5. Full documentation on all of the above posted the [the wiki](https://github.com/ioos/ioosngdac/wiki/Data-Set-Archiving)

kbailey-noaa, 2023-09-05:
(Note: Aaron and Kathy were testing the Glider DAC Project Board)

kerfoot, 2024-04-18:
Per email communication with John Relph (NCEI) on April 17, 2024:

`We normally check for new NGDAC data once a day. However, we have not seen any new data since June 2023.`

So it appears that the archival package creation is no longer functional. This should be addressed when time and funding permits.

benjwadams, 2024-04-18:
This has now been addressed and I will await feedback from NCEI.  We had to restore the Glider DAC server from a previous snapshot which had different configuration setup in the YAML files.  This setup was missing configuration variables for the archival script.  Going forward it'd be good to possibly ship out these archival logs and set up reasonable metrics for monitoring.

sarinamann-noaa, 2024-09-27:
Ben confirmed that this issue is still a priority after hurricane season

benjwadams, 2025-01-17:
Files have recently been written to archival.