ioos/glider-dac #235: Resolving registered user accounts

kerfoot, 2023-08-15:
There are 63 registered user accounts on the [account registration page](https://gliders.ioos.us/providers/admin), but only 36 submission directories at:

/data/data/priv_erddap

and only 36 submission directories at:

/data/submission

What is the best way to resolve this discrepancy?  I can contact the POCs for each registered account to figure out which ones are actually active.  Then, should we remove the user accounts that we confirm are not active?

benjwadams, 2024-04-18:
Now that we are in a more stable cycle of DAC development, I could send you over email addresses for users with nonexistent submission subdirectories.

benjwadams, 2025-03-26:
This is not a code issue, so removing from GH issues.  We can discuss this during our next meeting.

closed 2025-03-26 by benjwadams