ioos/glider-dac #236: Upgrade to ERDDAP v2.20

kerfoot, 2023-08-15:
ERDDAP 2.20 released today.  We should upgrade as soon as allowable.  

1. [v2.19 Changes ](https://coastwatch.pfeg.noaa.gov/erddap/download/changes.html)
2. [Upgrade Instructions](https://coastwatch.pfeg.noaa.gov/erddap/download/setup.html#update)

Though we [didn't upgrade to v2.19 as it was flawed](), there were also important changes in that version also:
1. [v2.18 Changes](https://coastwatch.pfeg.noaa.gov/erddap/download/changes.html#changes2.19)

benjwadams, 2023-10-20:
https://gliders.ioos.us/erddap/ was upgraded to 2.23, appears to work well other than minor cosmetic issues like broken question mark icons.   Closing due to upgrade exceeding v2.20.

kerfoot, 2023-10-26:
@benjwadams @Acolohan I'm reopening this until the "minor" broken icons are fixed.

benjwadams, 2023-12-06:
The icons have been fixed and we are currently running 2.23 via the docker-compose.yml declaration.  Closing as implemented.

closed 2023-12-06 by benjwadams