ioos/glider-dac #238: Process for modifying metadata on existing data sets

kerfoot, 2023-08-15:
Metadata on existing data sets is modified with the use of the extra_atts.json file. The file is created in the following directory:

**$PRIV_ERDDAP_ROOT/PROVIDER/DATASET_ID**

Here's an example of an extra_atts.json file to change the **NC_GLOBAL:id** and **temperature:standard_name**:

    {
        "_global_attrs": {
            "id": "mote-genie_20170901T175729Z"
        }
        "temperature": {
            "standard_name": "sea_water_temperature"
        }
    }

A few questions:
1. Once this file has been added with the appropriate changes, when/how are those changes made to the data set and reflected in the ERDDAP dataset and/or archived aggregated files?
2. Is a compliance check run on the data set once it has been updated? If so, when does this happen?
3. For previously NCEI archived data sets, how is NCEI notified of the changes so that they can update the accession?

This question is related to the following issues:
1. [issue 160](https://github.com/ioos/ioosngdac/issues/160)
2. [issue 159](https://github.com/ioos/ioosngdac/issues/159)

kerfoot, 2024-02-01:
As discussed on 2024-02-01 technical tag up, I ran into an issue with a previously archived data set ([g652-20230901T1200](https://gliders.ioos.us/status/datasets/?dataset_id=sg652-20230901T1200) in which the data provider used the incorrect WMO id.  

My understanding is that the <dataset /> snippet for a data set marked as Complete is not regenerated in the event that metadata is modified via submitting an extra_atts.json file. I would like to discuss implementing a watchdog that monitors all data sets marked as Complete. If an extra_atts.json file is uploaded by a data provider, the script would ideally do the following:

1. Notice the new extra_atts.json file
2. Place the data set in a queue that would recreate the ERDDAP <dataset /> element
3. Add the <dataset /> element to the ERDDAP datasets.xml
4. Wait for a major LoadDatasets event to reload the metadata
5. Creates an NCEI archival package and the associated md5 sum for NCEI pickup

This will allow for the modification of data set metadata and NCEI archiving under an existing accession number.

MathewBiddle, 2024-09-11:
Does this fix issues with datasets that can't currently be archived due to compliance issues?  ~170 datasets
 
For example:
```
  time: "status" is not a valid CF Standard Name Modifier
  time status: "flag" is not a valid CF Standard Name Modifier
  standard_name "time status flag" is not a valid CF Standard Name
```

benjwadams, 2025-11-14:
> 3. For previously NCEI archived data sets, how is NCEI notified of the changes so that they can update the accession?

NCEI pulls files and checks the MD5 sums.