ioos/glider-dac #239: Grafana notifications

kerfoot, 2023-08-15:
I keep getting multiple Grafana notification emails every day. 

A few questions:

1. Are these actual warnings that we need to be concerned about?
2. I am unable to log into the Grafana website to view details about the warnings. Can I get an account?

If the answer to either of the above is 'No', then I don't see a reason that I should be receiving the notifications and I would like to be removed from the distro list.

Acolohan, 2023-09-05:
@kerfoot is this still an issue?

kerfoot, 2023-09-05:
The notifications certainly don't help me. Not sure if they are helpful for RPS. I would like to be removed from all grafana email lists first as none of them provide any useful information to me. Once that is done, I'll close it. Not sure, but likely this is an @benjwadams task.

kerfoot, 2023-12-12:
@benjwadams can you remove me from the grafana email notifications?

benjwadams, 2024-03-27:
I don't receive notifications at that sort of frequency now.  There are a few useful alerts that can indicate when a glider deployment has not updated recently.   Would prefer to continue discussion of this issue via email if possible.

closed 2024-08-28 by sarinamann-noaa