ioos/glider-dac #240: NCEI authority tables and standardized vocabularies

kerfoot, 2023-08-15:
Tasks:

1. NCEI should be pushing updated authority tables to the [Glider DAC authority tables endpoint](https://gliders.ioos.us/ncei_authority_tables/), but the tables have not been updated since 2020.
2. Update current NCEI authority tables (institutions, projects, platforms, instruments)
3. Choose standardized vocabularies from:
    + [GCMD](https://www.earthdata.nasa.gov/learn/find-data/idn/gcmd-keywords)
    + [SeaDataNet](https://vocab.seadatanet.org/search)
    + [NERC](https://vocab.nerc.ac.uk/search_nvs/)

benjwadams, 2023-12-06:
Where does this stand as of right now?

leilabbb, 2024-09-06:
This is related to #234
 @benjwadams is the glider DAC pointing to or mirroring  https://www.ncei.noaa.gov/data/oceans/ncei/cfg/ngdac/

sarinamann-noaa, 2024-09-27:
Task #2 should be taken care of- will need to confirm with Jen B

sarinamann-noaa, 2024-10-24:
Task # 2 is completed

benjwadams, 2024-11-12:
Closing, authority table URLs now redirect to NCEI hosted URLs, and the Glider Compliance Checker Plugin has been updated accordingly.

closed 2024-11-12 by benjwadams