ioos/glider-dac #244: Dataset checksums

kwilcox, 2023-08-15:
Some deployments are missing the `ioos_dac_checksum` global attribute. For background on the attribute see https://github.com/ioos/ioosngdac/issues/58. Without the attribute it is impossible to know if the underlying files (data, qc, metadata attributes, etc.) have changed. The following deployments are affected:

* SG609-20140719T1700
* SG609-20150206T1720
* SG610-20140715T1400
* SG610-20150206T1750

flbahr, 2023-08-16:
Kyle, 
I don't see these under my deployments (mbari) login. I'm assuming this is the end of deployment check for the gliders? 
Who are these attributed to? If they were mine, I'd finalize them but I'm not seeing them when I'm in my deployments on the glider dac. 
Fred

benjwadams, 2023-09-13:
Closed as attribute is created by ERDDAP datasets.xml generation and should change when update is detected by changed timestamp for modification time.

closed 2023-09-13 by benjwadams