ioos/glider-dac #263: Remove/fix 2023 Navy glider duplicate datasets

kbailey-noaa, 2023-08-22:
Please update the GDAC to show only 1 deployment per Navy glider. 
@benjwadams is already addressing the issue that caused this: "plan is to disallow autoregistration for files, but when a new Navy Glider deployment is created, to pick up any already existing files and add them to the created deployment folder.  New files will then continue to be be picked up in the created deployment as usual."

...therefore, this GH issue requests a fix for the existing duplicates in the GDAC:  NG601, NG221 and NG309

Also - the two NG601 datasets cover different time ranges.  FYI, NG601 was actually deployed 8/3/2023.
There should be 1 NG601 dataset that covers 8/3 - present.

ng601-20230803T0000 - first profile: 8/9/2023; last profile: present day (8/22/2023)
ng601-20230803T155100 - first profile: 8/3/2023; last profile: 8/17/2023

kbailey-noaa, 2023-08-25:
@benjwadams Was autoregistration for files disallowed?

kerfoot, 2023-09-06:
The table below displays duplicate Navy glider data sets. After close inspection, the data sets (**bold**) that contain the seconds in the data set span the same time range and number of profiles.

**These 5 data sets have been deleted as they are duplicates**.

| dataset_id            | glider   |   wmo_id | start_date                | end_date                  |   lat_min |   lat_max |   lon_min |   lon_max |   num_profiles |   days |   latency_minutes |
|-----------------------|----------|----------|---------------------------|---------------------------|-----------|-----------|-----------|-----------|----------------|--------|-------------------|
| ng221-20230820T0000   | ng221    |  8901038 | 2023-08-20 23:28:49+00:00 | 2023-09-06 05:59:15+00:00 |   27.089  |   27.9345 |  -94.2883 |  -92.2683 |            261 |     17 |               569 |
| **ng221-20230820T224700** | ng221    |  8901038 | 2023-08-20 23:28:49+00:00 | 2023-09-06 05:59:15+00:00 |   27.089  |   27.9345 |  -94.2883 |  -92.2683 |            261 |     17 |               569 |
| ng230-20230820T0000   | ng230    |  4802934 | 2023-08-20 23:30:17+00:00 | 2023-09-06 06:11:20+00:00 |   27.0424 |   27.4105 |  -94.3312 |  -92.9654 |            184 |     17 |               557 |
| **ng230-20230820T224100** | ng230    |  4802934 | 2023-08-20 23:30:17+00:00 | 2023-09-06 06:11:20+00:00 |   27.0424 |   27.4105 |  -94.3312 |  -92.9654 |            184 |     17 |               557 |
| ng309-20230820T0000   | ng309    |  4802926 | 2023-08-20 23:08:14+00:00 | 2023-09-06 05:34:38+00:00 |   25.4192 |   27.4138 |  -95.8124 |  -94.0326 |            196 |     17 |               593 |
| **ng309-20230820T224300** | ng309    |  4802926 | 2023-08-20 23:08:14+00:00 | 2023-09-06 05:34:38+00:00 |   25.4192 |   27.4138 |  -95.8124 |  -94.0326 |            196 |     17 |               593 |
| ng601-20230803T0000   | ng601    |  8901039 | 2023-08-03 15:57:19+00:00 | 2023-09-06 09:07:48+00:00 |   26.8896 |   27.9994 |  -87.348  |  -84.3008 |            756 |     34 |               380 |
| **ng601-20230803T155100** | ng601    |  8901039 | 2023-08-03 15:57:19+00:00 | 2023-09-06 09:07:48+00:00 |   26.8896 |   27.9994 |  -87.348  |  -84.3008 |            756 |     34 |               380 |
| ng781-20230816T0000   | ng781    |  8901040 | 2023-08-16 22:18:39+00:00 | 2023-09-06 07:21:55+00:00 |   25.782  |   27.7564 |  -94.8889 |  -93.8455 |            299 |     21 |               486 |
| **ng781-20230816T215200** | ng781    |  8901040 | 2023-08-16 22:18:39+00:00 | 2023-09-06 07:21:55+00:00 |   25.782  |   27.7564 |  -94.8889 |  -93.8455 |            299 |     21 |               486 |

Closing once I confirm that the data sets have been removed from ERDDAP.

kerfoot, 2023-09-06:
Confirmed data sets removed. There are now 13 active NAVOCEANO glider data sets that are all updating as expected. Closing.