ioos/glider-dac #266: Security Updates for Glider Map

nguyandy, 2023-08-23:
Our DevOps Engineer, Pavani Ankam, has performed the following tasks last week for the EC2 instances that host Glider Map, and the data-catalog service.

- Install SSM Agent on EC2 instances to communicate with AWS Systems Manager so patches can be installed
- Run compliance status check against machine to check on what patches have been missing
- Install all dependencies needed for patching
- Install all important/critical security updates on machines to stay current with security updates
- Reboot the machine and use patch manager to run the final compliance check against machines. Confirm if they are compliant

kerfoot, 2023-09-07:
@nguyandy Can this issue be closed?

DonaldMoretti, 2023-09-13:
This is now completed

closed 2023-09-13 by DonaldMoretti