ioos/glider-dac #277: Implement aggregate QC flags to meet NDBC GTS Ingest requirements

kbailey-noaa, 2023-08-31:
NDBC currently does not read any glider QC flags as part of the GDAC harvest of real-time T and S data that they deliver to the GTS.  

In order for NDBC to read QC flags, the GDAC must implement the QARTOD “Aggregate/Rollup” flag variable, following the IOOS Metadata Profile requirements:
https://ioos.github.io/ioos-metadata/ioos-metadata-profile-v1-2.html#quality-controlqartod
and
https://ioos.github.io/ioos-metadata/ioos-metadata-profile-v1-2.html#requirements-for-the-qartod-aggregaterollup-flag

NDBC will only examine contents of the aggregate_quality_flag variable for filtering purposes for GTS harvest.  NDBC will not (and should not) read detailed QC flags.

> Rules governing the ‘Aggregate/Rollup’ flag variable:
> 
> The value of the variable is the UNESCO/QARTOD v2 convention:
> 9 = Missing
> 2 = Not Eval
> 1 = Pass
> 3 = Suspect
> 4 = Fail
> The value of the rollup flag should be the worst result out of all of the individual tests. It’s called a “Summary Flag” in the [QARTOD Data Flags](https://github.com/ioos/ioos_qc/blob/master/ioos_qc/qartod.py#L46-L77) manual (pg 3).
> Here’s how it’s done in the [ioos_qc library](https://github.com/ioos/ioos_qc/blob/3d8efba12644eb37b9ef35e6cdcf35189e789075/ioos_qc/qartod.py#L49-L80) with the [corresponding test](https://github.com/ioos/ioos_qc/blob/3d8efba12644eb37b9ef35e6cdcf35189e789075/tests/test_qartod.py#L1096-L1113).
> The variable should have a standard name attribute of aggregate_quality_flag. See the [QARTOD](https://ioos.github.io/ioos-metadata/ioos-metadata-profile-v1-2.html#quality-controlqartod) section for more information.
> NDBC should exclude any values that are QC fail (and missing) but include everything else (not eval, pass, suspect)

Please implement these aggregate flags at least for real-time T and S, as these are the only variables that NDBC is harvesting and delivering to the GTS.  

NDBC POCs: Dawn Petraitis and Bill Smith
IOOS DMAC POC: Micah Wengren

benjwadams, 2023-09-05:
Variable already exists in underlying NetCDF dataset, called `qartod_%(name)s_primary_flag`:  https://github.com/ioos/glider-dac/blob/main/glider_qc/glider_qc.py#L338-L366

Needs to be added to ERDDAP datasets.xml via modification of `scripts/build_erddap_catalog.py` script.

leilabbb, 2023-09-06:
The variable qartod_%(name)s_primary_flag does exist in the ERDDAP datasets.xml so there is no need to modify the build_erddap_catalog.py script

kerfoot, 2023-09-06:
I still do not understand what is going on with this. Here is the link to all data sets that have updated since September 1, 2023:

[Data Sets with min_time set to 2023-09-01](https://gliders.ioos.us/erddap/search/advanced.html?page=1&itemsPerPage=1000&searchFor=&protocol=%28ANY%29&cdm_data_type=%28ANY%29&institution=%28ANY%29&ioos_category=%28ANY%29&keywords=%28ANY%29&long_name=%28ANY%29&standard_name=%28ANY%29&variableName=%28ANY%29&maxLat=&minLon=&maxLon=&minLat=&minTime=2023-09-01&maxTime=)

49 data sets are returned. Now, if I leave the date as September 1, 2023 **AND** add that the data set must contain the **qartod_temperature_primary_flag**:

[Data Sets with min_time = 2023-09-01 AND contains qartod_tempreature_primary_flag](https://gliders.ioos.us/erddap/search/advanced.html?page=1&itemsPerPage=1000&searchFor=&protocol=%28ANY%29&cdm_data_type=%28ANY%29&institution=%28ANY%29&ioos_category=%28ANY%29&keywords=%28ANY%29&long_name=%28ANY%29&standard_name=%28ANY%29&variableName=qartod_temperature_primary_flag&maxLat=&minLon=&maxLon=&minLat=&minTime=2023-09-01&maxTime=)

The number of data sets returned goes from 49 to 1.

**Question**: why do the majority of real-time data sets not include the qartod* flags?

leilabbb, 2023-10-03:
Update on progress made:

- the glider_qc.py script is using the ioos-qc library to create QARTOD test results for geophysical / legacy variables [pressure, conductivity, temperature, salinity, density].

- the QARTOD rollup qc variable with the name "qartod_[legacy variable]_primary_flag" and standard_name "aggregate_quality_flag" is created for every legacy variable  existing in the netcdf glider file.

- the aggregate qc flags uses the ioos-qc the 'aggregate' and 'qartod_compare' methods to set the resultant rollup qc array of all tests run. The rollup up array takes the worst flag of a data point with the worst flag being 4 = FAIL in the following order: 
9 = Missing
2 = Not Eval
1 = Pass
3 = Suspect
4 = Fail

- The qc tests implemented and working are: 
'gross_range_test'  
'qc_rollup'

- The qc tests implemented and need more testing are:
'spike_test'
'rate_of_change_test'

- The qc tests implemented and not working is: 
'flat_line_test', 

see pull request #289

leilabbb, 2023-12-06:
To add to the above discussion, there is also the case of geophysical variables associated with incorrect standard names that need to be addressed to explain why QARTOD is not run on these variables.

leilabbb, 2024-02-06:
#277 should be marked as complete. 
Creating a new issue to address challenges encountered in implementing the QC test on the data arrays. See #318

leilabbb, 2024-02-13:
Yes.
 The aggregate QC flag is created for T and S unless the variable does not exist in the file or its array is all NaN or Fill Values.
 
**Example Variable "qartod_temperature_primary_flag"**
**In file "ru39_20231203T182905Z_rt.nc"** (

<class 'netCDF4._netCDF4.Variable'>
int8 qartod_temperature_primary_flag(time)
    valid_min: 1
    valid_max: 9
    _FillValue: 9
    units: 1
    flag_meanings: PASS NOT_EVALUATED SUSPECT FAIL MISSING
    flag_values: [1 2 3 4 9]
    ioos_category: Quality
    qartod_test: 'qc_rollup'
    standard_name: aggregate_quality_flag
    long_name: QARTOD Primary Flag for sea_water_temperature
    qartod_package: https://github.com/ioos/ioos_qc/blob/main/ioos_qc/qartod.py    
    references: http://gliders.ioos.us/static/pdf/Manual-for-QC-of-Glider-Data_05_09_16.pdf
    qartod_config: {'gross_range_test': {'suspect_span': [0, 35], 'fail_span': [-2, 40]}, 'spike_test': {'suspect_threshold': 
                               0.02396099641919136, 'fail_threshold': 0.04792199283838272}, 'rate_of_change_test': {'threshold': 
                               0.03594149462878704}, 'flat_line_test': {'tolerance': 1, 'suspect_threshold': 3600, 'fail_threshold': 
                               9000}}

**The values of the flags:**
masked_array(data=[1, --, --, 1, --, --, 1, --, 1, --, --, --, 1, --, 1,
                   --, --, 1, --, --, 1, --, 1, --, 1, --, --, 1, --, --,
                   --, 1, --, --, 1, --, --, 1, --, 1, --, --, 1, --, --,
                   --, --, 1, --, 1, --, --, 1, --, --, 1, --, 1, --, --,
                   1, --, --, --, 1, --, 1],
             mask=[False,  True,  True, False,  True,  True, False,  True,
                   False,  True,  True,  True, False,  True, False,  True,
                    True, False,  True,  True, False,  True, False,  True,
                   False,  True,  True, False,  True,  True,  True, False,
                    True,  True, False,  True,  True, False,  True, False,
                    True,  True, False,  True,  True,  True,  True, False,
                    True, False,  True,  True, False,  True,  True, False,
                    True, False,  True,  True, False,  True,  True,  True,
                   False,  True, False],
       fill_value=9,
            dtype=int8)

kbailey-noaa, 2024-03-13:
@leilabbb can you provide a link to a dataset where this is implemented?

leilabbb, 2024-08-22:
> @leilabbb can you provide a link to a dataset where this is implemented?

This [data set](https://gliders.ioos.us/erddap/tabledap/sp010-20150409T1524.html
) has the primary_flag implemented.

kbailey-noaa, 2024-08-23:
Thanks. 
@mwengren Can you pls review this? Does the variable name matter, or does NDBC just look at the standard name?

I thought we'd expect to see a variable name of, for example, sea_water_temperature_qc_agg but instead I see qartod_temperature_primary_flag.  But the standard name is aggregate_quality _flag...

leilabbb, 2024-08-28:
I think this was decided early on and was already in the system before I started working on QC.  There is this [document](https://docs.google.com/document/d/1rh5niGdqbPB0eIkJn9CO5fZmeknuHLrxVL0qBpoOEa8/edit) that has the *_primary_flag variable, which I believe won't break the GTS workflow.

mwengren, 2024-08-28:
The IOOS Metadata Profile [rules](https://ioos.github.io/ioos-metadata/ioos-metadata-profile-v1-2.html#requirements-for-the-qartod-aggregaterollup-flag) are based on using CF ancillary variables (so the data variable includes the name of the aggregate QC variable in its `ancillary_variables` attribute), and the aggregate variable also has standard name `aggregate_quality_flag`.  

At a quick glance, it looks like this is in line with those rules.  That should make it easier for NDBC to read Glider files in the same way they do the IOOS Metadata Profile files, but they'd have to confirm that probably.

kbailey-noaa, 2024-08-29:
Thanks all!
@leilabbb It sounds like we're ready to proceed with an NDBC test.  Can you pls email Bill Smith and Dawn Petraitis (cc: GDAC team) this dataset with a request to test?

leilabbb, 2024-08-29:
> data set

email sent

sarinamann-noaa, 2024-10-02:
@sarinamann-noaa will request feedback from Bill Smith and Dawn Petraitis- can schedule a meeting to get information on their needs

sarinamann-noaa, 2024-11-20:
---------- Forwarded message ---------
From: Dawn Petraitis - NOAA Federal <dawn.petraitis@noaa.gov>
Date: Wed, Nov 13, 2024 at 1:18 PM
Subject: Re: GDAC - Testing the use of QARTOD variables for GTS release
To: BaghdadBrahim, Leila <Leila.Baghdad-Brahim@tetratech.com>
Cc: Sarina Mann - NOAA Affiliate <sarina.mann@noaa.gov>, Bill Smith - NOAA Affiliate <bill.c.smith@noaa.gov>, Adams, Benjamin <benjamin.adams@tetratech.com>, Moretti, Donald <donald.moretti@tetratech.com>, Kathleen Bailey - NOAA Federal <kathleen.bailey@noaa.gov>

Hi Leila,
Thanks for the confirmation. I have submitted a ticket to have NDBC do the needed work to adjust our code. Since I just submitted the ticket, I don't have a date for when it will completed. Once I have a date, I'll let everyone know.

The good news is that we haven't seen any issues so far on our end with the addition of the QC aggregate flags to the real time data files.

Thanks,
Dawn

kerfoot, 2025-02-26:
I see that this issue has been marked as complete, but there are still many real-time data sets for which one or more of the *_primary_flag variables do not have the results of the tests in them, as noted in #408. Has this issue been marked as complete because it has been fixed or because there is a plan to fix it sometime in the future?

leilabbb, 2025-03-06:
The issue has been marked as complete because the aggregate QC flag has been implemented and the workflow with NDBC has been confirmed.

Datasets that are not showing the QARTOD flag are being investigated. In many cases, the glider_qc needs to be re-run on the files.

sarinamann-noaa, 2025-04-22:
--------- Forwarded message ---------
From: Bill Smith - NOAA Affiliate <bill.c.smith@noaa.gov>
Date: Fri, Apr 18, 2025 at 7:27 PM
Subject: Re: GDAC - Testing the use of QARTOD variables for GTS release
To: Kathleen Bailey - NOAA Federal <Kathleen.Bailey@noaa.gov>
Cc: BaghdadBrahim, Leila <Leila.Baghdad-Brahim@tetratech.com>, Dawn Petraitis - NOAA Federal <dawn.petraitis@noaa.gov>, Adams, Benjamin <benjamin.adams@tetratech.com>, Sarina Mann - NOAA Affiliate <sarina.mann@noaa.gov>, Moretti, Donald <donald.moretti@tetratech.com>, Jon Strauss - NOAA Affiliate <jon.h.strauss@noaa.gov>

The QARTOD QC aggregate flags have been incorporated into the GliderDAC harvesting software for NDBC.

--
regards,
Bill Smith

closed 2025-04-22 by sarinamann-noaa