ioos/glider-dac #286: Issue template for GitHub

benjwadams, 2023-09-13:
A template should be added for issues not pertaining to common usage problems that can be directed to the "ioos-tech" mailing list.

Right now, this is the text of what is showing up below the mailing list message:

> Don’t see your issue here? [Open a blank issue.](https://github.com/ioos/glider-dac/issues/new)

Also, I think the following section should be rephrased:

> Non-technical or code-related questions can be posed in 'ioos_tech' instead.

The first sentence could be misconstrued to mean that code issues are OK to post in the ioos_tech group, although I doubt many people will interpret it that way upon reading.  For clarity, I recommend the following sentence, or something similar:

> Questions not pertaining to technical or code related matters can be posed in 'ioos_tech' instead.

mwengren, 2023-09-13:
@benjwadams I think your rephrase makes sense.  Can you post a PR against this file?

https://github.com/ioos/.github/blob/main/.github/ISSUE_TEMPLATE/config.yml

This is where the ioos_tech link originates from across the IOOS repos.

The Glider DAC specific templates can be added as needed to this repo (I think they'll appear in order below the ioos_tech link).

benjwadams, 2023-09-14:
cc @kerfoot, @Acolohan, your input here regarding what should go in a base issue template for the Glider DAC would be appreciated.

kbailey-noaa, 2024-03-18:
@benjwadams I'd love to see the following templates:

- Bug report
- Documentation
- Feature request

This repo is a great example of what we should aim for: https://github.com/NOAA-EMC/WW3/issues/new/choose

cc: @mwengren, @MathewBiddle, @ocefpaf - I think we talked about this months ago so looping you in for awareness.

MathewBiddle, 2024-03-18:
LMK if you need help. I've become quite the proponent and user of templates. I think [this one](https://github.com/NOAA-PSL/CEFI-info-hub-list/issues/new?assignees=chiaweh2&labels=&projects=&template=add_cefi_resource.yml&title=%5BAdd+new+CEFI+resource%5D%3A+) is my most complex as it kicks off a [GitHub Action](https://github.com/NOAA-PSL/CEFI-info-hub-list/blob/main/.github/workflows/gha_generate_pr.yml) to do some work and submit a PR back to the repo.

ocefpaf, 2024-03-28:
BTW, those 3 items are, IMO, what we should have in all IOOS repo. Should we do that in the `.github` repo to get them automagically in all repos? Note that, the individual repos can override the default we write in `.github` if they want/need to.

sarinamann-noaa, 2024-10-02:
Request @mwengren close this issue if work has been completed

closed 2024-10-02 by mwengren