ioos/glider-dac #291: Develop a plan for OG1.0 Implementation

Acolohan, 2023-10-04:
Need to begin the conversation about this

kbailey-noaa, 2024-02-27:
@kerfoot Can you pls provide a status update of this plan?

kerfoot, 2024-02-27:
Willl do asap. I'm putting together my UG2 talk on OG1.0 for March 15 and this plan is central to that. Should have a broad plan finished soon that we can then discuss with the DAC team and broader DMAC community.

kimmotikka, 2024-08-19:
OG1.0 was published last June. What is the current status of this issue?

leilabbb, 2024-08-19:
This is in progress and an update will follow soon.

leilabbb, 2024-08-30:
To start implementing OG1.0 into the GDAC, we need to compare two documents: [NGDAC NetCDF File Format Version 2] (https://ioos.github.io/glider-dac/ngdac-netcdf-file-format-version-2.html) and the [OceanGliders 1.0 format](https://github.com/OceanGlidersCommunity/OG-format-user-manual/blob/main/OG_Format.adoc). We'll discuss any differences and additions, focusing on how they might impact current processes and the time needed for testing and validation.

sarinamann-noaa, 2025-02-10:
User request- Victor Turpin (vturpin@wmo.int) would benefit from OG1.0 implementation to resolve issues with different vocabularies from GDAC, Oceanglider, Ocean Ops. He is undertaking a larger project to derive metadata without the manual download of NetCDFs.

ocefpaf, 2025-02-10:
There is a prototype for an OG compliance-checker plugin in https://github.com/uw-farlab/cc-plugin-og.

sarinamann-noaa, 2026-01-16:
---------- Forwarded message ---------
From: Kathleen Bailey - NOAA Federal <Kathleen.Bailey@noaa.gov>
Date: Thu, Jan 15, 2026 at 7:28 AM
Subject: OG1.0 at VIMS

GDAC Team - changing the subject line but this is in response to the VIMS exchange below.

Flagging this comment from D, that "we are using pyglider to generate the CF-compliant netCDF files from the raw glider data files, then updating those files to OG1.0 standard ourselves." If they've already implemented OG1.0 it doesn't make sense for us to reinvent the wheel. We should just leverage their code.  I'm not sure where we are with OG1.0 but can you pursue this?  And if we can use their code it would be fantastic if they could share it on our Glider-DAC Github repo. 

This is an example of where we can work more closely with the community to efficiently accomplish goals.

Thanks,
Kathy

---------------------------------------------------------

From: D
Date: Wednesday, January 14, 2026 at 12:34 PM

Hi Leila,

For the glider netCDF data files, we are using pyglider to generate the CF-compliant netCDF files from the raw glider data files, then updating those files to OG1.0 standard ourselves. Some of the fields in the OG1.0 specification asks for URL and DOI. Since we don't know what the URL for the data will be yet, we plan to leave them as N/A. Do you prefer that we handle them differently?

Thanks