ioos/glider-dac #294: Conda deactivate environment hangs

kerfoot, 2023-10-17:
I am able to activate a conda environment in order to run some python scripts. However, when I deactivate:

    > $ conda activate gdac_status
    (gdac_status) conda deactivate

The terminal hangs and never properly exits from the environment. This has not always happened, so my guess is something has changed on the server side.

Any ideas @benjwadams ?

benjwadams, 2023-10-17:
I'm not a huge conda user, @ocefpaf may know more.  If you could get some verbose output, that'd be helpful for diagnosing the issue.

ocefpaf, 2023-10-18:
It can be the shell configuration. Hard to debug without access to the system. I'm not sure that is even caused by conda itself, `conda deactivate` is just a shell `source` that restore the environment variables from before the activation.

PS: we can post it on the [conda discourse](https://conda.discourse.group/) to see if other experienced this.

kerfoot, 2023-12-11:
closing this as I no longer have access to either the prod or dev server, so it doesn't matter anymore.

closed 2023-12-11 by kerfoot