ioos/glider-dac #297: Hurricanes-2023 dataset showing up in ERDDAP, maps

kbailey-noaa, 2023-10-23:
Map-based tools (e.g. [IOOS Model Viewer](https://oceansmap.com/link/WphOjcW5Q4y8SjI92Meb5A)) are showing the Navy glider hurricanes-2023 dataset and I see it's back in [ERDDAP](https://gliders.ioos.us/erddap/tabledap/hurricanes-20230601T0000.html).  Last time (July 2023) we saw this, @benjwadams found:

"Hurricanes-2023 is being picked up outside of the directory it is supposed to be present in.  I’ll see if I can stop it from being picked up by the deployments directory.  This is one reason I really wanted Navy Gliders directory for submission to reside outside of /data/ mount, as it’s a “meta” directory that only really serves to symlink to other directories due to Navy’s differences in upload process from the usual spec."

Ben, can you please make this change again?

kerfoot, 2023-10-23:
I don't see any deployments on the map as of 2023-10-23 10:17EDT

nguyandy, 2023-10-23:
@kerfoot deployments should be on the map now. The status trajectory json files weren't accessible earlier but @benjwadams resolved it.

kerfoot, 2023-10-23:
Not quite resolved yet....

There are tracks, but there are 2 deployments (hurricanes*) that are being served via erddap and these are displayed as bad tracks.  The hurricanes* deployments need to be removed from datasets.xml

[image]

kbailey-noaa, 2023-10-23:
This was resolved.  Thanks!

closed 2023-10-23 by kbailey-noaa