ioos/glider-dac #312: What are the files in the gh-pages branch _nc directory.

ocefpaf, 2023-12-21:
@kbailey-noaa and @MathewBiddle last call we raised the question regarding the files under `_nc` folder. They are all from the `ioosngdac` repository but not all of them are used in the pages.

| file                                                                 | used in                                              |                                              |
| -------------------------------------------------------------------- | ---------------------------------------------------- | -------------------------------------------- |
| \_nc/template/IOOS_Glider_NetCDF_v2.0.cdl                            | \_docs/glider-background-and-sampling-terminology.md | \_docs/ngdac-netcdf-file-format-version-2.md |
| \_nc/template/IOOS_Glider_NetCDF_v2.0.nc                             | \_docs/glider-background-and-sampling-terminology.md | \_docs/ngdac-netcdf-file-format-version-2.md |
| \_nc/template/IOOS_Glider_NetCDF_v2.0.ncml                           | \_docs/glider-background-and-sampling-terminology.md | \_docs/ngdac-netcdf-file-format-version-2.md |
| \_nc/template/IOOS_Glider_NetCDF_v3.0-noqartod.cdl                   | ❌                                                   |                                              |
| \_nc/template/IOOS_Glider_NetCDF_v3.0-noqartod.nc                    | ❌                                                   |                                              |
| \_nc/template/IOOS_Glider_NetCDF_v3.0-noqartod.ncml                  | ❌                                                   |                                              |
| \_nc/template/IOOS_Glider_NetCDF_v3.0-qartod.cdl                     | ❌                                                   |                                              |
| \_nc/template/IOOS_Glider_NetCDF_v3.0-qartod.nc                      | ❌                                                   |                                              |
| \_nc/template/IOOS_Glider_NetCDF_v3.0-qartod.ncml                    | ❌                                                   |                                              |
| \_nc/template/additional_variables/cdom.cdl                          | ❌                                                   |                                              |
| \_nc/template/additional_variables/chlorophyll_a.cdl                 | ❌                                                   |                                              |
| \_nc/template/additional_variables/dissolved_oxygen.cdl              | ❌                                                   |                                              |
| \_nc/template/additional_variables/geophysical_variable_template.cdl | ❌                                                   |                                              |
| \_nc/template/additional_variables/instrument_template.cdl           | ❌                                                   |                                              |
| \_nc/examples/profile/ru29-20140101T0942.nc                          | ❌                                                   |                                              |
| \_nc/examples/profile/ru30-20140702T2335.nc                          | ❌                                                   |                                              |

Only three files are mentioned in the docs. The rest of them are missing some context and/or are left over from previous docs iterations. Should we remove them?

kbailey-noaa, 2024-01-18:
@kerfoot @benjwadams  Can you please weigh in on the status of these various v3.0 and variable files that aren't referenced anywhere in our Documentation pages? They are 7+ years old.  Should they be removed?

kerfoot, 2024-01-18:
These are example NetCDF and CDL files used by providers as a guide to create their own NetCDF files. I would prefer that they are not deleted as they provide guidance for new users, but you can delete them if you feel it is necessary.

ocefpaf, 2024-01-20:
What if:

1. We delete the mc files to safe space and avoid hosting binaries.
2. Document what/how/when to use those CDL that generate the nc files.

Does that make sense? My guess is that, looking at this from the point of view of a newcomer to the glider world, most would not know what to with them or even how to find them.

leilabbb, 2024-09-03:
@ocefpaf is there a reason we are keeping this under a separate branch? The templates for example should be moved to the main branch for easy find. Please let us know your thoughts.

sarinamann-noaa, 2025-03-17:
@ocefpaf can you please provide your thoughts as per Leila's previous comment?

ocefpaf, 2025-03-17:
Sorry, somehow I missed Leila's message here. The separate branch here is quite special, it was a merge from another repo, the ngdac, that was made to consolidate things into a single repo. If anything from here makes sense to be moved to the main branch, it is not for me to say. I'd ask Kathy or someone on the glider-dac team. I can help with the tech part of the page building and repo history.

PS: My recommendation would be to not move anything though. Unless we are re-build these pages from the ground up.

benjwadams, 2025-03-26:
I don't know that I'm in agreement with keeping docs on a separate branch.  If we have a feature branch that necessitates a change in the documentation, we'll need to update the docs later upon merge, whereas if the `main` branch had the docs, future feature branches could update them as part of the PR.

ocefpaf, 2025-03-26:
We can bring the docs to the main branch. It won't be simple, but doable. A counter point is to imagine this docs branch as another repo. fork from it, merge to it, etc. It is not an uncommon setup.

closed 2026-05-06 by leilabbb