ioos/glider-dac #313: netCDF files submission:  A User Technical Guide.

leilabbb, 2024-01-03:
The attached document is intended to walk you through the technical steps of creating a deployment and submitting your netCDF files. Complying with the workflow is important to help the Gilder team resolve any issues you might encounter in a timely manner.  

[Wokflow_ncFile_Submission.docx](https://github.com/ioos/glider-dac/files/13815442/Wokflow_ncFile_Submission.docx)

sarinamann-noaa, 2024-09-03:
Needs to move to the Wiki but we need a team discussion on that process

sarinamann-noaa, 2024-10-09:
Will bring up with full team at next meeting

sarinamann-noaa, 2025-04-08:
This documentation should live in this location so users can access it:

https://ioos.github.io/glider-dac/ngdac-netcdf-file-submission-process.html

leilabbb, 2026-10-01:
Closing issue: ngdac-netcdf-file-submission-process.md has been added to the gh-pages repository, reviewed, and the document is now part of the wiki pages.

closed 2026-10-01 by leilabbb