ioos/glider-dac #318: address challenges encountered in implementing the QC test on the data arrays #144

leilabbb, 2024-02-06:
This issue was created based on findings from working on #277

The QC process needs additional conditional statements to handle special cases of data arrays. The following have been already identified and are being tested:

Check the time array for invalid timestamps, duplicate timestamps, and non-ascending timestamps
Check the legacy variable array for an empty array, an array with all fill values, or nans
Check the legacy variables' standard names for non-CF compliant names
Check variables sharing the same standard name as the legacy variables

In the above cases, the QC tests are skipped and the QARTOD variable is filled with FLAG 2: NOT_EVALUATED, and the dac_comment attribute is annotated with failing test results.

leilabbb, 2024-02-14:
**Remark**: need a consistent way to parse the deployment start time to complete test 1.

leilabbb, 2024-08-14:
The following tests have been completed:

- [ x] Test 1: Check the time array
- [ x] Test 2: Check the legacy variables
- [ x] Test 3: Check  the attribute "standard_name"
- [ x] Test 4: Check the attribute "ancillary_variables"
- [x ] Test 5: Check shared "standard_name" attributes

leilabbb, 2024-08-23:
The following test need to be added:

- [ ] Test 6:

Include non-zero timestamps that are invalid in the timestamps check.
Action 1:
The QC tests are skipped and the Global attribute dac_qc_comment is annotated with test 6 results.

sarinamann-noaa, 2024-10-02:
@leilabbb will verify the last test

closed 2025-01-29 by leilabbb