ioos/glider-dac #319: GDAC map date/time settings

kbailey-noaa, 2024-02-13:
@nguyandy 
I can't set a date in the GDAC map.  I changed the custom range to 8/15/23 - 9/15/23 but then I can't set the map time.  I'm unable to even choose any year other than 2024.

Custom range setting:
[image]

Map date/time setting:
[image]

Also - can you add 2023 to the custom range?
[image]

nguyandy, 2024-02-13:
Will look at this today!

kbailey-noaa, 2024-02-14:
@nguyandy It looks like it was corrected / looks good now. I can select custom range and I see a full range of years to choose from instead of just 2024.  I also see the 2023 button was added.  Thanks!