ioos/glider-dac #322: Navy glider data flow update

kbailey-noaa, 2024-02-27:
Pls work with the Navy reps (POC: Danny Illich) to update the Navy glider data flow.

The Navy data flow upgrade involved a costly 'workaround' solution of pushing all Navy gliders' files to 1 single folder.  We requested they separate the deployment files into respective deployment folders, as the GDAC system was designed to handle.  

The individual glider directories are already created when Tony Gilless registers/creates the deployment so we're just asking that they push the files to those deployment directories.  

Bill Lingsch is taking the lead and coordinated a meeting with Navy reps and the GDAC team to review this process.

Acolohan, 2024-02-28:
Just adding Ben and John for this. The meeting is at 0900 on 29 Feb.

kbailey-noaa, 2024-03-08:
Bill Lingsch's meeting notes: [IOOS_Navy GDAC Meeting Feb 29 2024.docx](https://github.com/ioos/glider-dac/files/14541245/IOOS_Navy.GDAC.Meeting.Feb.29.2024.docx)

**Action Items:**
1.	John Kerfoot provide Navy with the older Pearl Scripts to Danny Illich by March 11th.
2.	Danny Illich makes a cursory review of the code to see if it is useable within their environment by March 18th.
3.	NAVO will provide Bill Lingsch with a follow on time to meet given their tight schedules and GDAC team will flex to that.  
4.	Bill Lingsch sets up a follow on technical meeting to detail the way ahead to get to the end state and associate responsible parties.

Acolohan, 2024-05-01:
Need Provider assistance on this.

kbailey-noaa, 2024-08-08:
This is paused during hurricane season, and will be pushed again in ~Dec 2024.

sarinamann-noaa, 2025-02-04:
Moving to backlog as per @kbailey-noaa takeaways from discussion meeting

benjwadams, 2025-04-03:
Communications made to NAVOCEANO indicate that this is unlikely to happen in the foreseeable future, despite the apparent simplicity of the changes.   Going to close as not planned.  If NAVOCEANO modifies their workflow, it should be relatively easy to accommodate and won't require additional changes, but we may want to revert the single folder code.

benjwadams, 2025-04-09:
Moving to backlog.

benjwadams, 2025-11-04:
As of right now, NAVOCEANO is unable to modify the requests and we have not been able to obtain the processing code.  This is outside of our control, so I am going to mark this as not planned, but have hopes that discussion can resume in the future.

closed 2025-11-04 by benjwadams