ioos/glider-dac #323: Develop web application for the curation of data set metadata by data providers

kerfoot, 2024-02-28:
Per this morning's phone discussion, we propose to close and split #223 into 2 separate tasks:

1. Development of a web application for adding & modifying metadata to DAC data sets by the data providers. Presently, this can only be accomplished via the use of a json file as outlined in #238. While this mechanism does serve its intended purpose, json syntax errors can lead to issues. 
2. Development of a "Raw Data Upload Tool" that leverages the metadata application defined above. This tool, which has periodically been discussed over the years, would allow data providers to fill out metadata forms and submit raw glider data files to the DAC. The DAC would then combine the supplied metadata with the raw data files, process the files to the [DAC specification](https://ioos.github.io/glider-dac/ngdac-netcdf-file-format-version-2.html), register the deployment and upload the files to appropriate location.

This issue focuses on the first item in this list. Data providers would log in, supply the requested metadata and save the records. These records could then subsequently be modified and the changes would be applied to the existing data set files, providing an more highly curated and complete data set. The metadata could be modified by either DAC administrators or the data provider, depending on the scale of the modifications desired. It would also eliminate the resubmission of previously submitted NetCDF files to modify existing data sets.

kbailey-noaa, 2024-03-08:
The GDAC community should be the primary source for new GDAC features or additions.  Please identify a user(s) who are seeking this feature, and they should also be considered a tester.  Will set status to icebox.

benjwadams, 2024-04-18:
As metadata is often submitted with errors, this is an important feature to have on future development roadmaps.  We do have a mechanism right now to address this by submitting a JSON file, but having a user interface accessible through the providers application and persisting changes to the database would be better in the long run from a usability perspective.

benjwadams, 2025-03-26:
Data Upload tool status is somewhat in limbo at this point, but work may resume on it as project priorities dictate.

For metadata modifications, we will likely extend the functionality of the existing Providers application rather than creating a new web application.

Hence, I am closing this in favor of moving discussion to https://github.com/ioos/glider-dac/issues/420.

closed 2025-03-26 by benjwadams