ioos/glider-dac #324: Development of a Raw Data Upload Tool

kerfoot, 2024-02-28:
Per this morning's phone discussion, we propose to close and split #223 into 2 separate tasks:

1. Development of a web application for adding & modifying metadata to DAC data sets by the data providers. Presently, this can only be accomplished via the use of a json file as outlined in #238. While this mechanism does serve its intended purpose, json syntax errors can lead to issues. Details filed under #323 
2. Development of a "Raw Data Upload Tool" that leverages the metadata application defined above. This tool, which has periodically been discussed over the years, would allow data providers to fill out metadata forms and submit raw glider data files to the DAC. The DAC would then combine the supplied metadata with the raw data files, process the files to the [DAC specification](https://ioos.github.io/glider-dac/ngdac-netcdf-file-format-version-2.html), register the deployment and upload the files to appropriate location.

This issue focuses on the second item in this list. Upon completion of #323, this metadata model and record could then be used to process raw glider NetCDF files to the [DAC specification](https://ioos.github.io/glider-dac/ngdac-netcdf-file-format-version-2.html) and apply the supplied metadata to the files. 

The intent of this tool is to lower some of the technical barriers associated with metadata models and NetCDF file creation, hopefully leading to increased participation for smaller organizations using gliders in their research. Given that this issues has been discussed for quite some time, I would also propose getting feedback from the user community on the value of this type of application before deciding on whether to proceed.

Regardless of whether this issue is addressed, #323 must be completed as it will provide a much simpler and easier way to modify data set metadata by the data providers.

kbailey-noaa, 2024-03-08:
#223 will be closed, and this Raw Data Upload tool will be set aside (Icebox) until a user is identified.

sarinamann-noaa, 2024-10-02:
@leilabbb did not identify a user at the UG2 Workshop for this issue. This can be closed and revisited if a user is identified in the future

closed 2024-10-02 by sarinamann-noaa