ioos/glider-dac #335: Sverdrup-20200925T1200 issue

kbailey-noaa, 2024-04-10:
There are 2 datasets that appear to have overlapping data:
[Sverdrup-20190926T0000](https://gliders.ioos.us/erddap/tabledap/Sverdrup-20190926T0000.html) -- 9/26/2019 to 1/5/2020
[Sverdrup-20200925T1200](https://gliders.ioos.us/erddap/tabledap/Sverdrup-20200925T1200.html) -- 9/26/2019 to 10/17/2020

Sverdrup-20200925T1200 - I assume the dataset was supposed to start 9/25/2020 but the first value is 9/26/2019. I looked at both datasets for 9/26/2019 and the data are the same: [Sverdrup-20190926T0000](https://gliders.ioos.us/erddap/tabledap/Sverdrup-20190926T0000.htmlTable?trajectory%2Cwmo_id%2Cprofile_id%2Ctime%2Clatitude%2Clongitude%2Cdepth%2Cconductivity%2Ccrs%2Cdensity%2Cinstrument_ctd%2Clat_uv%2Clon_uv%2Cplatform_meta%2Cprecise_lat%2Cprecise_lon%2Cprecise_time%2Cpressure%2Csalinity%2Csource_file%2Ctemperature%2Ctime_uv%2Cu%2Cv&time%3E=2019-09-26T00%3A00%3A00Z&time%3C=2019-09-27T12%3A46%3A19Z) and [Sverdrup-20200925T1200](https://gliders.ioos.us/erddap/tabledap/Sverdrup-20200925T1200.htmlTable?trajectory%2Cwmo_id%2Cprofile_id%2Ctime%2Clatitude%2Clongitude%2Cdepth%2Cconductivity%2Ccrs%2Cdensity%2Cinstrument_ctd%2Clat_uv%2Clon_uv%2Cplatform_meta%2Cprecise_lat%2Cprecise_lon%2Cprecise_time%2Cpressure%2Csalinity%2Csource_file%2Ctemperature%2Ctime_uv%2Cu%2Cv&time%3E=2019-09-26T00%3A00%3A00Z&time%3C=2019-09-26T22%3A28%3A25Z).

It seems Sverdrup-20200925T1200 contains both deployments in its dataset.

Can someone please take a look?

kerfoot, 2024-04-11:
looking into this now. Will fix/update asap.

kerfoot, 2024-04-12:
I removed all of the *2019*.nc files from this data set and am waiting for the ERDDAP data set to reflect the changes.

@benjwadams since this data set is marked as complete, will the changes in the time coverage cause the NCEI archival package to be re-created and a new md5 generated?

I also noticed an issue with [non-unique profile_id values](https://gliders.ioos.us/erddap/tabledap/Sverdrup-20190926T0000.htmlTable?profile_id%2Clatitude%2Clongitude&distinct()) in [Sverdrup-20190926T0000](https://gliders.ioos.us/status/datasets/?dataset_id=Sverdrup-20190926T0000). I sent an email explaining both of these issues to the data provider and steps that need to be taken in order to fix the issue.

kerfoot, 2024-04-12:
@benjwadams Although I removed the [Sverdrup-20200925T1200 bad source files](https://gliders.ioos.us/providers/users/gcoos_dmac/deployment/5f6f7c4a003f14001768cd6e) a couple of hours ago, the [Sverdrup-20200925T1200 data set](https://gliders.ioos.us/erddap/tabledap/Sverdrup-20200925T1200.graph?longitude,latitude,profile_id&longitude%3E=-97.529&longitude%3C=-91.527&latitude%3E=24.539&latitude%3C=30.542&.draw=markers&.marker=5%7C5&.color=0x000000&.colorBar=%7C%7C%7C%7C%7C&.bgColor=0xffccccff&.zoom=out) has not updated yet, despite multiple ERDDAP reload data set processes having run.

Can you check into this and let us know why the time range still shows data from both [Sverdrup-20200925T1200](https://gliders.ioos.us/erddap/tabledap/Sverdrup-20200925T1200.graph?longitude,latitude,profile_id&longitude%3E=-97.529&longitude%3C=-91.527&latitude%3E=24.539&latitude%3C=30.542&.draw=markers&.marker=5%7C5&.color=0x000000&.colorBar=%7C%7C%7C%7C%7C&.bgColor=0xffccccff&.zoom=out) and [Sverdrup-20190926T0000](https://gliders.ioos.us/erddap/tabledap/Sverdrup-20190926T0000.graph?longitude%2Clatitude%2Cprofile_id&.draw=markers&.marker=5%7C5&.color=0x000000&.colorBar=%7C%7C%7C%7C%7C&.bgColor=0xffccccff) in the [Sverdrup-20200925T1200](https://gliders.ioos.us/erddap/tabledap/Sverdrup-20200925T1200.graph?longitude,latitude,profile_id&longitude%3E=-97.529&longitude%3C=-91.527&latitude%3E=24.539&latitude%3C=30.542&.draw=markers&.marker=5%7C5&.color=0x000000&.colorBar=%7C%7C%7C%7C%7C&.bgColor=0xffccccff&.zoom=out) data set?

kerfoot, 2024-04-12:
Done. The 2 data sets now accurately reflect the actual time window of the deployments.

closed 2024-04-12 by kerfoot