ioos/glider-dac #342: Update Glider DAC application architecture

benjwadams, 2024-04-24:
1) Update Glider DAC to use the application factory pattern: https://flask.palletsprojects.com/en/2.2.x/patterns/appfactories/
This makes testing and refactoring the application much easier.

2) Move away from MongoDB and BerkeleyDB to a relational database by using SQLAlchemy.

These are already well underway on this branch:
 https://github.com/benjwadams/glider-dac/tree/application_factory_fixes_rebase
1) Is basically finished at this point.
2) Is well underway but needs further testing and migration scripts need to be reinstated.

sarinamann-noaa, 2024-09-03:
Will discuss to determine if this fits within new BIL

sarinamann-noaa, 2024-10-09:
Priority after hurricane season- upgrade slated to take place and resolve this

benjwadams, 2024-12-11:
Related to #346

sarinamann-noaa, 2025-03-17:
@benjwadams what is the status of this issue, with [#346 ](https://github.com/ioos/glider-dac/pull/346) merged?

benjwadams, 2025-04-03:
These changes are already present on https://github.com/ioos/glider-dac/tree/develop .
I'd really like to close out this issue but can't really count it as complete until that is merged into `main`, and it will also likely be another major release.

closed 2025-04-03 by benjwadams