ioos/glider-dac #344: Intermittent errors when requesting data from OSU685-20231023T0000

ocefpaf, 2024-05-01:
The [OSU685-20231023T0000](https://gliders.ioos.us/erddap/tabledap/OSU685-20231023T0000.html) seems to be load intermittently. Sometimes all of its data is there, sometimes it is incomplete sometimes it is gone alltogeteher. See the image below where I captured a state where nothing was there.

[image]

@benjwadams is this a known issue? If so please feel free to close this.

benjwadams, 2024-05-07:
That's odd.  Could you put an approximate time stamp on a time where you attempted to load that dataset and encountered an error so that I can check logs?

ocefpaf, 2024-05-08:
I got these based on my slack communication with the DMAC team when it happened:
 
- April 18th 19:08
- April 30th 19:13

But it did happen more than twice.

benjwadams, 2024-07-17:
I had odd issues with this deployment where it was looping within the watchdog re-applying QC to the one of the same files in the deployment.  A recent commit has fixed this, so unless this occurs again, can I close this out?

closed 2024-07-17 by benjwadams