ioos/glider-dac #347: Unsafe hash upon account creation

scalzava, 2024-05-15:
To whom it may concern.

Our security team is working on the automated detection of session vulnerabilities in opensource web applications, including insecure hashing of authentication credentials. Our analyzer identified that the set function of glider_util/bdb.py is using an unsafe hashing function (SHA-256) to store users' passwords. This practice might leave your application vulnerable to offline bruteforcing attacks. Please see the OWASP recommendations for secure password hashing: https://cheatsheetseries.owasp.org/cheatsheets/Password_Storage_Cheat_Sheet.html

Can you take a look into the relevant code parts and comment on the issue?

DonaldMoretti, 2024-05-15:
We do not use SHA-256

closed 2024-05-15 by DonaldMoretti

benjwadams, 2024-05-15:
Specifically we have SHA512 from passlib, with iterations and salt.  Not sure how this originated, but we possibly have better algorithms we could use.