ioos/glider-dac #349: Quick access to list of WMO IDs and associated calls signs

kbailey-noaa, 2024-05-22:
Navy gliders loaned for hurricane glider missions are sometimes new to the GDAC, and sometimes not.  As a result, the Navy isn't sure if they need to request a new WMO ID.  Requesting a way to generate a list that is user-friendly, and post a link to it on our User Documentation.

benjwadams, 2024-05-28:
I think this information should be displayed on the main page as well as the deployment detail page.  It's a very small amount of information to reside in a table or webpage, and there's really no reason not to display it.

kbailey-noaa, 2024-07-17:
@benjwadams I'm looking for a list of glider call signs and associated WMO IDs.  It should be up to date (i.e. dynamic, so the list is always current) and the list should be publicly accessible.

closed 2024-10-02 by kbailey-noaa