ioos/glider-dac #350: GDAC Map issues

kbailey-noaa, 2024-05-28:
1) Several of the annual GDAC tracks have issues with huge lines crossing the map, I assume due to erroneous glider locations in the dataset -- almost every year 2020 and older.  Screenshot below because per #2 the permalink doesn't work.  

[image]

2) When I selected the year 2018 per above via the calendar icon and I created a [permalink](https://gliders.ioos.us/map/?permalink=%7B%22l%22%3A%7B%22geojson%22%3A%5B%22gliderdac_collection_3%22%2C%22gliderdac_collection_0%22%2C%22gliderdac_collection_2%22%2C%22gliderdac_collection_1%22%2C%22gliderdac_collection_7%22%2C%22gliderdac_collection_8%22%2C%22gliderdac_collection_4%22%2C%22gliderdac_collection_5%22%2C%22gliderdac_collection_6%22%5D%7D%2C%22m%22%3A%7B%22b%22%3A%5B-228.94868831523587%2C-38.64529312384428%2C25.75834293476419%2C68.0925302630552%5D%7D%7D), it did not retain the timeframe of 2018.  Tracks shown are present day.

benjwadams, 2024-05-29:
These erroneous tracks all appear to be going to lat/lon 0, 0, no?  Probably a simple way would be to filter these all out that go there.  More complex would be to actually perform a QARTOD location test, although if we had repeated 0, 0 coordinates, it might possibly not filter out on its own if I'm remembering the QARTOD tests correctly.

nguyandy, 2024-05-29:
> When I selected the year 2018 per above via the calendar icon and I created a [...]

OM1 permalinks do not retain time range data, so this is expected behavior. Do we want to add this as a new feature?

kbailey-noaa, 2024-05-29:
@nguyandy I was thinking about how the time range is retained in OM3 permalinks, and perhaps we need to reconsider the GDAC map and upgrade to OM3 or reconsider some features.  It hasn't changed much in several years.

leilabbb, 2024-07-24:
To evaluate the glider track, we will undertake the following steps:

1- Apply the QARTOD location test to ensure accurate geographical positioning.
2- Utilize the test-generated flags to plot the tracks on the map.

leilabbb, 2024-07-26:
Details for geographical positioning QC process:

- [ ] 1- Apply the QARTOD location test to ensure accurate geographical positioning. 

The values used to plot the tracks on the map come from individual data points submitted in the raw profile files under the variables **profile_lat** or **profile_lon**.  Note that the QARTOD location test does not apply to individual data points. We can apply the QARTOD location test to the lat and lon arrays in the raw file and then compare the results with the profile_lat and profile_lon values. 

- [ ] 2- Utilize the test-generated flags to plot the tracks on the map.

We can add a flag to the profile_lat,lon to help filter out erroneous values

Here is an example from a raw data file "ng1116-20240717T155900.nc" to show the variables:

The **profile_lat** has an unreasonable value of 13.363155. In comparison, the measured variable **lat** contained reasonable values [min:26.1, max=26.3]. 

> Variable "lat"
> In file "ng1116-20240717T155900.nc"
> double **lat**(time=355);
>   :_FillValue = -999.0; // double
>   :ancillary_variables = "lat_qc";
>   **:comment = "Values may be interpolated between measured GPS fixes";**
>   :coordinate_reference_frame = "urn:ogc:crs:EPSG::4326";
>   :long_name = "Latitude";
>   **:observation_type = "measured";**
>   :platform = "platform";
>   :reference = "WGS84";
>   :standard_name = "latitude";
>   :units = "degrees_north";
>   :valid_max = 90.0; // double
>   :valid_min = -90.0; // double 
> **Scalar variable has values: [ nan, min:26.1, max=26.3]**.      <--- Reasonable

> Variable "profile_lat"
> In file "ng1116-20240717T155900.nc"
> double **profile_lat**;
>   :_FillValue = -999.0; // double
>   **:comment = "Value is interpolated to provide an estimate of the latitude at the mid-point of the profile";**
>   :long_name = "Profile Center Latitude";
>   **:observation_type = "calculated";**
>   :platform = "platform";
>   :standard_name = "latitude";
>   :units = "degrees_north";
>   :valid_max = 90.0; // double
>   :valid_min = -90.0; // double
>  
> **Scalar variable has value: 13.363155**.    <--- Unreasonable

leilabbb, 2024-08-01:
> Details for geographical positioning QC process:
>
> * [ ]  1- Apply the QARTOD location test to ensure accurate geographical positioning.
>
> The values used to plot the tracks on the map come from [...]

The QARTOD location test needs a bounding box defined by (min lon, min lat, max lon, max lat) to constrain the geographical area of the glider mission. However, the providers' files lack this information. Using a random bounding box won’t necessarily address the glider track outliers and shouldn’t be considered a reliable solution.

A temporary solution will be to check the profile_lat/lon used in generating the track array against the midpoint value of the measured lat and lon of the glider profile.  This functionality should be added to [glider_qc.py](https://github.com/ioos/glider-dac/blob/main/glider_qc/glider_qc.py#L361)

```
with Dataset(profile_file, '+r') as nc:
    '''
    Verify the track's lon and lat coordinates.
    ''' 
    profile_lat = nc.variables['profile_lat'][:]
    profile_lon = nc.variables['profile_lon'][:] 
    lat = nc.variables['lat'][:]
    lon = nc.variables['lon'][:]

   # Calculate the number of standard deviation the profile_lat/lon exceeds the measured mean
    num_std_lat = np.abs((profile_lat - np.mean(lat)) / np.std(lat))
    num_std_lon = np.abs((profile_lon - np.mean(lon)) / np.std(lon))
   
   # check the number of standard deviations and replace bad values with None
    if num_std_lat > 3 or num_std_lon > 3:
        nc.variables['profile_lat'][:] = None
        nc.variables['profile_lon'][:] = None
   
```
> 
> * [ ]  2- Utilize the test-generated flags to plot the tracks on the map.
> 
> We can add a flag to the profile_lat,lon to help filter out erroneous values

There is no need for a flag the function in [trajectories.py](https://github.com/benjwadams/glider-dac-status/blob/plots_traj_hotfix/status/trajectories.py#L72) should eliminate the values = None

> Here is an example from a raw data file "ng1116-20240717T155900.nc" to show the variables:
>
> The **profile_lat** has an unreasonable value of 13.363155. In comparison, the measured variable **lat** [...]

kbailey-noaa, 2024-08-02:
@leilabbb can this qc test be expedited? It's becoming more urgent.

@benjwadams 
@DonaldMoretti

kbailey-noaa, 2024-08-05:
@benjwadams Also, can you pls scrub the latest files of the bad GPS locations?

leilabbb, 2024-08-07:
The [qc test](https://github.com/ioos/glider-dac/commit/fe13c23a517797864aa261665a1054ae4e67a007) is ready under review

sarinamann-noaa, 2024-09-03:
Other issues open that directly deal with this.

closed 2024-09-03 by sarinamann-noaa