ioos/glider-dac #363: remove the duplicated entries of QARTOD variables under the ancillary_variables attributes

leilabbb, 2024-08-13:
The ancillary_variables attribute under the variables being QCed is listing duplicates of QARTOD variables. See an example file following this datasets link: https://gliders.ioos.us/erddap/tabledap/ng309-20230820T0000.html

Here is a variable example:
> temperature {
>     Float32 _FillValue -999.0;
>     Float32 actual_range 5.4466, 32.0449;
>     String ancillary_variables "_temperature_qc qartod_temperature_flat_line_flag qartod_temperature_gross_range_flag qartod_temperature_rate_of_change_flag qartod_temperature_spike_flag qartod_temperature_primary_flag_ **qartod_temperature_flat_line_flag qartod_temperature_gross_range_flag qartod_temperature_rate_of_change_flag qartod_temperature_spike_flag qartod_temperature_primary_flag** _qartod_temperature_flat_line_flag qartod_temperature_gross_range_flag qartod_temperature_rate_of_change_flag qartod_temperature_spike_flag qartod_temperature_primary_flag_ **qartod_temperature_flat_line_flag qartod_temperature_gross_range_flag qartod_temperature_rate_of_change_flag qartod_temperature_spike_flag qartod_temperature_primary_flag**";
>     Float64 colorBarMaximum 32.0;
>     Float64 colorBarMinimum 0.0;
>     String instrument "instrument_ctd";
>     String ioos_category "Temperature";
>     String long_name "Sea Water Temperature";
>     String observation_type "measured";
>     String platform "platform";
>     String standard_name "sea_water_temperature";
>     String units "degrees_C";
>     Float32 valid_max 40.0;
>     Float32 valid_min -5.0;
>   }

A quick ERDDAP searches show more than 200 datasets with similar issues: 

174 datasets:
https://gliders.ioos.us/erddap/search/index.html?page=1&itemsPerPage=1000&searchFor=-delayed+qartod+%22qartod_density_flat_line_flag+qartod_density_gross_range_flag+qartod_density_rate_of_change_flag+qartod_density_spike_flag+qartod_density_primary_flag+qartod_density_flat_line_flag%22

86 datasets:
https://gliders.ioos.us/erddap/search/index.html?page=1&itemsPerPage=1000&searchFor=delayed+qartod+%22qartod_density_flat_line_flag+qartod_density_gross_range_flag+qartod_density_rate_of_change_flag+qartod_density_spike_flag+qartod_density_primary_flag+qartod_density_flat_line_flag%22

sarinamann-noaa, 2024-10-02:
@leilabbb will link pull request and check ERDDAP to close the issue

leilabbb, 2025-01-29:
The dataset [ng309-20230820T0000](https://gliders.ioos.us/erddap/tabledap/ng309-20230820T0000.html) is **FIXED**: no duplicated entries of QARTOD variables under the ancillary_variables attributes are present, however other datasets are still showing this issue.

sarinamann-noaa, 2025-03-17:
@leilabbb any update on this issue?

leilabbb, 2025-03-18:
> [@leilabbb](https://github.com/leilabbb) any update on this issue?

Investigating when the system will be back up, but in some cases, manual intervention may be necessary.

leilabbb, 2025-04-08:
The issue ' duplicated entries of QARTOD variables under the ancillary_variables ' still exists: [SG677-20210924T1605](https://gliders.ioos.us/erddap/tabledap/ng309-20230820T0000.html)

leilabbb, 2025-11-26:
[SG677-20210924T1605](https://gliders.ioos.us/erddap/tabledap/ng309-20230820T0000.html) Fixed

[sp034-20210505T2105](https://gliders.ioos.us/erddap/tabledap/sp034-20210505T2105.html) Not Fixed

The issue is till present, opening a new system wide investigation.

closed 2026-11-05 by benjwadams