ioos/glider-dac #364: remove old QARTOD variables from delayed mode datasets

leilabbb, 2024-08-13:
This is related to issue #355 

The fix in PR #360 took care of the recently submitted delayed mode datasets and any future datasets.  

The remaining delayed data sets need a filter to delete the old GDAC QARTOD variables.

This ERDDAP search shows the listing of the datasets in question:
 https://gliders.ioos.us/erddap/search/advanced.html?page=1&itemsPerPage=1000&searchFor=delayed+qartod+-temperature_qartod_&protocol=%28ANY%29&cdm_data_type=%28ANY%29&institution=%28ANY%29&ioos_category=%28ANY%29&keywords=%28ANY%29&long_name=%28ANY%29&standard_name=%28ANY%29&variableName=%28ANY%29&maxLat=&minLon=&maxLon=&minLat=&minTime=&maxTime=

leilabbb, 2024-12-18:
This needs a manual intervention to delete and reprocess the delayed mode data sets.

sarinamann-noaa, 2025-12-10:
as per GDAC team meeting 12/20/25:
Requires global check on dataset- need to determine if there are still old QARTOD variables. Would require some manual intervention. 
Not a low-effort task.